riscv-non-isa/riscv-trace-spec #229: How does the instruction trace interface report cm.popret when data trace is implemented?

j0106223, 2025-04-26:
In Section 4.3 (Data Trace Interface requirements), there is a description of how to handle instructions that involve multiple memory accesses. However, I would like to clarify a few details.

It is mentioned that these multiple memory access activities must be reported on the instruction trace interface. What exactly does 'reporting' these activities mean?

Does it require a visible activity report, meaning iretire != 0 and itype != 0?

Or is it acceptable to treat these multiple memory accesses as 'no activity' (iretire = 0 and itype = 0), just like how they are handled before data trace is mentioned, since that would still conform to itype = 0?

If reporting them as "no activity" is not acceptable, then referring to the example of cm.popret in Section 4.3, are the following expected values for iretire and iaddr correct?

A cm.popret is equivalent to 4 loads and a return:

1. Block - last instruction is cm.popret, itype = 0 (data trace interface reports 1st load);
2. Single - cm.popret, itype = 0 (data trace interface reports 2nd load);
3. Single - cm.popret, itype = 0 (data trace interface reports 3rd load);
4. Single - cm.popret, itype = 13 (data trace interface reports 4th load);

My expected instruction trace interface would show the following signal activity:

1. iretire != 0 and itype = 0, iaddr = pc (1st instruction retired in this block);
2. iretire = 1 and itype = 0, iaddr = pc(cm.popret);
3. iretire = 1 and itype = 0, iaddr = pc(cm.popret);
4. iretire = 1 and itype = 13, iaddr = pc(cm.popret).

Thank you for your clarification!

IainCRobertson, 2025-04-29:
Jack,
         See in-line...

Iain

[Iain] Yes, but no.  iretire !=0 only.  itype can be 0

[Iain] No, because the load/store interface signals provide additional load/store information about retiring instructions.  The encoder still needs to know which instruction the load/store details belong to.  This is needed for various reasons (ensuring load/store trace is properly synchronized with the instruction trace stream for example, or if encoder-based filtering is enabled, it will decide whether to generate data trace packets based on whether the load/store instruction itself matches the filtering criteria.

[Iain] Yes, this is correct

j0106223, 2025-04-29:
Hi Iain,

Thanks for your detailed reply — it clarified my question.
I also noticed a typo in my original message. I meant to ask:

`Does it require a visible activity report, meaning iretire != 0 and itype == 0?`

Thanks again for pointing that out!

Best regards,
Jack

closed 2025-04-30 by j0106223