riscv-non-isa/riscv-trace-spec #238: Clarify Whether CBO.ZERO and AMO Are Considered Multi-Memory-Access Instructions

j0106223, 2025-07-01:
Hi Iain,

When data trace is implemented:

Should CBO.ZERO and AMO be considered multi-memory-access instructions ?
Both of these instructions perform multiple memory accesses:

A CBO.ZERO instruction performs multiple stores of zero to all bytes within the targeted cache block.

> A cbo.zero instruction performs stores of zeros to the full set of bytes corresponding to the cache block
> whose effective address is the base address specified in rs1.

An AMO instruction typically issues one load and one store as part of its operation.

> The atomic memory operation (AMO) instructions perform read-modify-write operations for
> multiprocessor synchronization and are encoded with an R-type instruction format. These AMO
> instructions atomically load a data value from the address in rs1, place the value into register rd, apply
> a binary operator to the loaded value and the original value in rs2, then store the result back to the
> original address in rs1.

If they are considered multi-memory-access instructions,
should each memory access be reported on the instruction trace interface?

Best regards,
Jack

j0106223, 2025-07-01:
Hi Iain,

But AMO load and store operations are merged and reported as a single activity (dretire) on the data trace interface.

> Atomic and CSR accesses have either both load and store data, or store data and an operand. For CSRs and unified atomics, both values are reported via data, with the store data in the LSBs and the load data or operand in the MSBs.

In this case, is it correct to report the load and store separately as multi-memory-access instructions on the instruction trace interface?

Best regards,
Jack

IainCRobertson, 2025-07-01:
Jack,
        There is no provision in the standard for reporting atomics as a discrete load/store pair, so I can't say it's "correct".  I believe it could work, by which I mean that a decoder could reconstruct the data accesses and properly correlate them with the instructions, provided it was knew that the hart is splitting up atomics like this.  But it would be non-standard.

Iain

j0106223, 2025-07-01:
Hi Iain,

What’s the standard-compliant way to report atomics?

Should they be reported only once on the instruction trace interface?
If that’s the case, does it mean AMOs are not actually considered multi-memory-access instructions?

Best regards,
Jack

closed 2025-07-09 by j0106223